pbatard/uefi-ntfs #3: UEFI:NTFS does not work correctly on Asus Z87 mainboard

steve6375, 2016-01-23:
I tried using Rufus on various USB Flash drives by formatting as NTFS using UEFI:NTFS option.
I also manually created NTFS+FAT32 primary partitions on both Removable and Fixed disk USB drives with same result (using Easeus Home Partition Master).

The Z87 firmware can directly UEFI boot from the NTFS partition - so no problem there as it seems to have built-in NTFS drivers.
However, if I choose the second formware UEFI boot option, it boots to the Rufus EFI files and then says it cannot find an NTFS partition.
The drives do boot via Rufus on Vbox so it looks like a Asus Z87 firmware compatibility issue.
I just wondered what the firmware was doing to cause the issue.

pbatard, 2016-01-24:
Hi Steve - thanks for the report.

Can you copy/paste the full output from UEFI:NTFS? There should be an error code + message after "cannot find an NTFS partition", and I'd like to know what it is.

I suspect that the issue has to do with the UEFI firmware embedding its own NTFS driver. In that case, when we attempt to mount the NTFS partition un UEFI:NTFS, we are probably using the ASUS embedded driver, and this may be the cause of the issue.

At any rate, UEFI:NTFS was designed to work even if the firmware provides an NTFS driver, because in that case, the first option that will pop out is the one using the native firmware driver, so i don't really see a major problem in having the second option not working - it's not needed since the firmware can already boot NTFS, and I don't believe many people (especially ones that are unfamiliar with the whole thing) will start by choosing the second option. And since it'll boot successfully with the first, noone should ever get stuck on that step.

steve6375, 2016-01-25:
Sorry for late reply. Here is screenshot.
As you say, in my case, I can boot using the native NTFS support, but ideally the EFI drive should also work (or not attempt to load in addition to the Asus NTFS driver). I was also concerned that other (non-NTFS UEFI) systems may also have a similar issue.
[image]

pbatard, 2016-01-25:
> in my case, I can boot using the native NTFS support

That will __always__ be the case for UEFI firmwares that have an NTFS driver. As I explain, UEFI:NTFS is not needed then, so there is no possibility that someone who has NTFS support in the firmware will ever be left stranded by UEFI:NTFS. And, as long as you follow the guidelines and place the UEFI:NTFS partition at the end (which you have to do for REMOVABLE devices anway, as you'll never be able to access the NTFS partition from Windows otherwise), the first option will always be the one using the firmware NTFS driver, so, again, I this should never be a major issue for users.

That doesn't mean I won't try to address it, but I do not think it is that big a deal.

> ideally the EFI drive should also work

Either that, or it should produce an error message that says "Try the other option".
Not sitting on a multi-million dollar lab with tons of UEFI motherboards I can play with, this might be tricky to address, but I'll see what I can do. I believe I should be able to get my hands back on an Intel NUC, that I know also has a native NTFS driver, and see if I can reproduce the same issue.
I'm hoping it will, as it would greatly simplify my investigations...

Otherwise, I guess we'll have to try to extract the NTFS driver from the ASUS Z87 firmware and test it on a UEFI system (either as a replacement for the UEFI:NTFS NTFS driver, to see what happens, or by loading it in an UEFI shell and then see if we can replicate the issue). But for that I need to know the exact model/version of your UEFI firmware (a download link would be nice if you have one). On a side note, one thing I've been wondering for a long time (but never got around to look into) is how difficult it would be to __inject__ an NTFS EFI driver in an UEFI firmware to make that system support native NTFS booting.

At the moment, the one way I see to try to address the issue would be to add a feature in UEFI:NTFS to try to detect if an NTFS driver is already provided by the UEFI system (not completely sure this is doable without attempting to mount the NTFS partition... which could create all kind of problems) and either:
1. Try to unload (if doable) it so that we use our own driver
2. Display a halt message that states "You shouldn't need UEFI:NTFS - Use the other boot option" (but then if the driver is not working for whatever reason, this could leave users stranded).
3. Try to use it (but then I'd like to be able to test that it works... which I should be able to do on the NUC)

steve6375, 2016-01-25:
If you can repro it and find an easy fix then no problem. Otherwise, just leave it. I just reported it for your interest really as I wondered if it would also happeb on a non-NTFS UEFI system.

steve6375, 2016-01-25:
The Z87-A firmware is the latest version from the Asus site.

pbatard, 2016-01-26:
Got my hands on the NUC and was able to replicate the issue :smiley_cat: 

I have now applied changes that will detect if the NTFS partition is already being serviced by a File System driver (`EFI_OPEN_PROTOCOL_TEST_PROTOCOL`), and will only load the provided driver if that's not the case.

Tests on the NUC as well as on an UEFI machine that does not come with a native NTFS EFI driver now seem to behave as expected (boot works regardless of the partition selected).

commit e0c944bc (2016-01-26): handle NTFS partition being serviced by a native driver

* Some UEFI firmwares (NUC, Asus and some AMI derivatives) include a native NTFS EFI driver.
  Detect this to prevent a fatal error on ConnectController()
* Also use OpenProtocol() instead of HandleProtocol() as per Phoenix' recommendations:
  http://wiki.phoenix.com/wiki/index.php/EFI_BOOT_SERVICES#HandleProtocol.28.29
* Closes #3
closed 2016-01-26 by pbatard