pbatard/uefi-ntfs #7: I can't find step-by-step instructions on how to create a UEFI:NTFS USB stick

SetTrend, 2016-08-03:
I need to create a USB stick containing two partitions:

1. **FAT**, containing NTFS drivers and boot code to launch autorun.inf on USB stick partition 2.
1. **NTFS**, containing the contents for the current Windows 10 ISO file

How can I get this to work? I can't find instructions on how this is done with Rufus.

pbatard, 2016-08-03:
1. Select your Windows 10 ISO.
2. Set "Partion scheme and target system type" to "GPT partition scheme for UEFI"
3. Set "File system" to NTFS

That's it.

If that doesn't answer your question, feel free to reopen this issue.

closed 2016-08-03 by pbatard

SetTrend, 2016-08-05:
This doesn't seem to work. Rufus only creates an NTFS, but not a FAT32 partition:

[image]

pbatard, 2016-08-05:
1. Be mindful that the UEFI:NTFS partition is so small that DIsc Management might not display it.
2. Please provide the FULL LOG of how you created your drive. It will be very explicit from it whether the UEFI:NTFS partition was created. Just click on the Log button from Rufus.

Also, be mindful that, if your flash drive is of type removable (99% of flash drives out there), you will NEVER be able to access the FAT partition from Windows, because will only ever mount the first partition with a file system it can service from a removable drive.