pbatard/uefi-ntfs #8: Install of Server 2016 STD hangs on spinning dots

TBSteck, 2017-01-10:
I created a bootable USB drive using Rufus v2.11.995 with the Windows Server 2016 STD .ISO file.  The install starts, the gray bar at the bottom goes across twice, like normal, the Windows Icon appears then only a 1/2 circle of dots (3 or 4 dots) show and then the install hangs.  This is on a Intel certified server brand new trying to do the initial install of Server 2016 with NVMe drives, therefore the boot has to be UEFI to recognize the drives ... assuming I ever get that far.

Intel support has given up on a solution and has sent it to engineering, but that could take a lifetime to get an answer from them.

Could it be something with the creation of the USB stick?  I've tried it on two different similar configured brand new servers (just CPU and memory qty are different). If I use a straight FAT32 (w/Windows 10 install - Server 2016 install has one file > 4 GB so can't use FAT32 USB) the install will complete successfully.

pbatard, 2017-01-10:
> Windows 10 install - Server 2016 install has one file > 4 GB

Interesting (and you're right, the `install.wim` of `en_windows_server_2016_x64_dvd_9327751.iso` is indeed larger than 4GB)... I have to wonder how Microsoft will play that, as, until now they had limited ISOs with >4GB files to evaluation versions, and not releases, and I don't think there's an official Microsoft tool that handles the creation of bootable USB media from such ISOs.

> If I use a straight FAT32 the install will complete successfully.

I seem to recall that the Windows installation process isn't always happy to find another NTFS partition, especially if is on a FIXED drive.

Does the USB drive you use have the FIXED atribute by any chance? You can find that from the Rufus log when it reports your drive properties with something like:
```
Disk type: Fixed, Sector Size: 512 bytes
```
If the drive doesn't have the FIXED attribute, you will see something like:
```
Disk type: Fixed, Sector Size: 512 bytes
```
You may also want to check whether your sector size is 512 bytes, as you might be using a 4K sector based disk, which usually spells trouble for boot.

Still, if it were me, I'd ask Microsoft support how one is supposed to convert `en_windows_server_2016_x64_dvd_9327751.iso` for UEFI USB installation, given that FAT32 is not usable. They'll probably tell you that you're supposed to split the WIM file, but I'd still would like to hear what they have to say...

pbatard, 2017-01-11:
For the record, on an intel NUC, I was able to get to the target partition selection screen using when USB drive created with Rufus from the `en_windows_server_2016_x64_dvd_9327751.iso` image.
I can confirm that the boot process successfully went through UEFI:NTFS before the Windows installer was launched.

Now, since I don't want to overwrite my partition, I didn't go through the actual installation, but, from what I understand, your install hanged before the target partition selection screen...

TBSteck, 2017-01-11:
I just finished another test ...

I took a brand new workstation that I just built (we are also a system builder) and took the RuFus USB stick and put the workstation in UEFI boot mode and installed Server 2016 STD and then took that SATA drive, put it back in the Hot Swap cage and the server booted up (using UEFI) until it got to the spinning dots and it hung, just like the install did.

Conclusion: There is nothing wrong with the RUFUs creation and the physical USB is not the problem either.  It has to do with the Intel's UEFI booting process.

My next test is to complete the install on the server using a Legacy (non UEFI) boot using the SATA SSD (Can't boot on the NVMe drives under UEFI).  I'm pretty sure this will install and boot just fine but that doesn't help my client which wants to boot and run on the NVMe SSD drives.  We have 3 servers here for my client and can't install until Intel fixes or figures out the problem.

Tim Steck
TBS Consultants
ClientCare@TBSConsultants.com<mailto:ClientCare@TBSConsultants.com>
(541) 767-2100

pbatard, 2017-01-11:
I agree with your assessment that Rufus or UEFI:NTFS have nothing to do with your problem. I will close this issue then.

closed 2017-01-11 by pbatard