pbatard/uefi-ntfs #10: Confused on installation steps

davidm71, 2017-08-20:
Hi,

I installed VS 2017 and was able to compile UEFI-NTFS to point of QEMU loading the driver in the virtual window. What steps do I need to do next to prepare my usb stick with Rufus? Thanks.

PS: Also can I put more than one efi driver on the usb stick?

pbatard, 2017-08-20:
You do know that, if you run Rufus in advanced mode, it will give you the possibility to install a __blank__ UEFI:NTFS onto your USB drive, right?

[image]

You must first click on the white triangle to get the Advanced Options, but once you have done that, you'll find that you can install a series of blank bootloaders, that you cannot insall in regular mode, including UEFI:NTFS.

If all you want is to create an USB with an empty NTFS partition (where you'll copy your content), and that can boot through UEFI, you don't have to bother recompiling and installing your own version of UEFI:NTFS.

Now, if you need more on how the FAT partition that Rufus installs is being created, please see https://github.com/pbatard/rufus/tree/master/res/uefi for information

I'm afraid that's about all the help I can provide though, as I consider that someone who wants to install UEFI:NTFS should have enough expertise to be able to create an USB Flash Drive with 2 partitions, and how to setup UEFI for boot (especially if they take a good look at the `uefi-ntfs.img` from Rufus.

> PS: Also can I put more than one efi driver on the usb stick?

Sure can. If you open the current `uefi-ntfs.img` from Rufus with anything that can read a FAT image (e.g. 7-zip) you'll see that it includes the drivers (and boot loaders) for x86_64, x86_32 and ARM, which means UEFI:NTFS can be used on any of these platforms. I'm also planning to add ARM64 support at some stage (probably shortly before Microsoft starts to release the Windows on ARM64 version they are currently keeping under wraps).

davidm71, 2017-08-20:
Thank you so much for the in depth reply. I sure hope I have the technical expertise because I'm trying to make a multi-Os boot stick. Got a gpt drive with Clover efi partition, two Osx hfs partitions, followed by a Windows2Go ntfs partition. Only problem is cant boot the Windows to go partition at all so hoping your Uefi-ntfs loader will do the trick. Thanks again.

davidm71, 2017-08-21:
Stupid question: Which resource file in Rufus that I'm viewing with 7-zip houses the uefi-ntfs.imf??

Thanks

davidm71, 2017-08-21:
Hey Pete,
Is it possible to mod your UEFI:NTFS Bootloader to boot Windows from another volume on another separate physical disk? Kind of like an emergency boot loader??

Thanks

pbatard, 2017-08-21:
> Is it possible to mod your UEFI:NTFS Bootloader to boot Windows from another volume on another separate physical disk?

I don't see why not. Now, if you are asking if I have any plans of ever doing something like this, the answer is no: you'll have to do it yourself by forking the project and adding whichever feature you want (like [this person did to add multiboot support](https://github.com/linnaea/uefi-ntfs-multiboot)).

closed 2017-09-07 by pbatard