pbatard/uefi-ntfs #13: Could not locate 'efi\boot\bootx64.efi

Kounex, 2018-07-27:
[image]

[image]

[image]

I can't figure out why i get this message. I would appreciate any kind of help!

Best
Kounex

pbatard, 2018-07-27:
> I can't figure out why i get this message

I can. A standalone UEFI:NTFS drive __will not boot anything__ (there is a reason why standalone UEFI:NTFS is only displayed when showing the advanced device options: it is meant for people who know what they are doing). You have to provide EFI bootloaders on the NTFS partition (__which is different from the FAT32 partition used by UEFI:NTFS where you will find the OTHER `bootx64.efi` bootloader).

If you don't know what all this means, then you should not use the UEFI:NTFS option. What exactly are you trying to accomplish here?

Also, you shouldn't report a Rufus issue in the UEFI:NTFS project. If you are using Rufus, then you should create an issue in Rufus (and follow the checklist there).

Kounex, 2018-07-27:
The UEFI:NTFS partition has been created in the process of creating a bootable image which exceeds the FAT32 size limit. I assumed the message regarding missing `bootx64.efi`, even though it is located in the UEFI:NTFS partition, might come up because of the NTFS service which is already running at this point. 

Thank you for pointing out that the file is actually missing in the FAT32 partition. As stated i thought this would be a UEFI:NTFS problem because Rufus created everything as intended.

closed 2018-07-27 by Kounex

pbatard, 2018-07-27:
> Thank you for pointing out that the file is actually missing in the FAT32 partition.

It's the other way round. The file is present on the FAT partition (the `UEFI:NTFS` one), but were missing on the NTFS partition (the one you are trying to boot).

Now, of course, if you used a 32-bit ISO that does not provide a `bootx64.efi` and then try to boot the USB on a 64-bit UEFI system, it will complain that it can't find `bootx64.efi` on the NTFS partition, since only a `bootia32.efi` will be there.

This is something that can usually be very easily validated by providing a complete Rufus log of how you created the image.

But anyway, I'm glad you sorted it out on your own.