pbatard/uefi-ntfs #17: Stuck on "Starting NTFS partition service" (HP Envy x2 / ARM64)

wbenny, 2018-10-20:
I've created bootable USB with Windows 10 (ARM64) with Rufus 3.3, plugged it to HP Envy x2 (with Snapdragon 835), choose to boot from USB, it printed few messages, the last one being `Starting NTFS partition service`. Then it's stuck. Any idea where the problem could be?

wbenny, 2018-10-20:
ISO is downloaded from Microsoft website (part of MSDN subscription), the filename: `18247.1001.180922-2224.rs_prerelease_CLIENT_CONSUMER_A64FRE_en-us`, sha256: `d7a31202f14f559c31f628349c1ca88dda4d55db62500d09e3917de8f0982e4c`, image attached. 
[image]

wbenny, 2018-10-20:
If you'd like me to provide any other information, just tell :)

pbatard, 2018-10-20:
Thanks for the report. I suspect this is an issue with the HP UEFI firmware (HP has a long history of being very temperamental when it comes to UEFI). But if you're lucky, it may just be an issue with the NTFS driver. So can you please try to replace the `EFI\Rufus\ntfs_aa64.efi` you'll find on the UEFI_NTFS partition created by Rufus, with the one you can find [here](https://rufus-web.akeo.ie/testing/ntfs_aa64.efi), and let me know if that works any better?

wbenny, 2018-10-20:
After bit of searching, it really looks like there's something wrong with HP's UEFI - I found on their forum people complaining that they aren't able to boot linux there, for instance. Also, that ARM64 ISO will boot if the USB drive is formatted as FAT32 - unfortunatelly, it'll throw BSOD after a while (early on, still when the "boot circle" is spinning). This is something that people were complaining about as well.

That being said, I'm running out of ideas how to perform clean ISO installation on this device :)

Anyway, will try and will let you know! Thank you.

wbenny, 2018-10-21:
BUT... if I start the `ntfs_aa64.efi` "manually" first, and then run "manually" the `bootaa64.efi`, the output is like this:

[image]

wbenny, 2018-10-21:
Eventually, if you have several possible ideas and you want to try them out (i.e. with various code paths taken/debug messages printed), I'm open to accepting bunch of `ntfs_aa64.efi` experimental files and trying them out for you :)

pbatard, 2018-10-21:
>  if I run the `ntfs_aa64.efi` "manually" first, and then run "manually" the `bootaa64.efi` (...)

That's the expected result. Not sure I have much of any ideas, especially as I validated that the `ntfs_aa64.efi` from Rufus 3.3 appears to be fine on real ARM64 hardware, and I also have had reports from other people with ARM64 hardware that were able to use UEFI:NTFS just fine.

So I'm afraid you appear to be in a minority where UEFI:NTFS doesn't seem to work, which I suspect has everything to do with the HP UEFI firmware. You may want to contact HP support to ask them why their firmware doesn't work whereas the one from other manufacturers does.

wbenny, 2018-10-21:
Alright, fair enough. I had this weird itch from the beginning that the issue would be on HP side, but still have had some hope. :) Thanks for your time!

closed 2018-10-21 by wbenny

pbatard, 2018-12-03:
If you feel like testing, you can try UEFI:NTFS from Rufus 3.4, that has a slightly updated version of the NTFS driver compared to 3.3. A BETA version of Rufus 3.4 can be found [here](https://rufus.ie/downloads/).

I don't expect it to change much about this issue, but it's worth a try I guess.

branzo, 2018-12-04:
@pbatard could the BETA2 be used as well?

branzo, 2018-12-04:
I already tried to get in touch with HP on this matter as i wanted to re-install Windows because of a b0rked update. 

Still no answer from them.. and i'm starting to consider to fill up for a refund. 

Thanks anyway.
Keep up the awesome work! ^,^

paulohuerta, 2022-12-05:
> Criei USB inicializável com Windows 10 (ARM64) com Rufus 3.3, conectei ao HP Envy x2 (com Snapdragon 835), optei por inicializar a partir do USB, imprimiu algumas mensagens, sendo a última `Starting NTFS partition service`. Então está preso. Alguma ideia de onde pode estar o problema?

Hi! managed to solve your problem, I have the same problem, I'm starting to believe that it is a problem with the manufacturer of my notebook, which in this case is Samsung