pbatard/uefi-ntfs #18: UEFI boot issues

brookspeppin, 2019-01-03:
I'm having a bunch of issues getting laptops to boot UEFI using standard Windows 10 install ISOs.  It shows in the boot list, and after it is selected the system attempts to boot and then bombs out and flips back to HD. Secure boot is off. Tested on Dell XPS 13 and Dell E7470. Using Rufus 3.4.1430.

I've also tried on "removable" and "Fixed disk" usb drives as well as 1803, 1809 win10 ISOs.

pbatard, 2019-01-03:
I need a bit more details. When UEFI:NTFS fails, it will freeze and show you the complete output of what it's been trying to do, with a clear error message.

So I need to see that. Unless you can provide that, then I cannot help you, as you may want to understand that, as a standalone Free Software developer, I don't have the means to access any of the systems you mention for testing.

pbatard, 2019-01-03:
For the record, here is the type of output you should see from UEFI:NTFS if your UEFI firmware was launching it properly:

[image]

Again, if you don't even see the first line, then it means that the issue is upstream to UEFI:NTFS and therefore not something I can do anything about.

brookspeppin, 2019-01-03:
Bummer. As far as I know Dell uses standard UEFI boot mechanisms. I can successfully UEFI boot using standard UEFI boot on USB (using fat32), but it just doesn't work with UEFI:NTFS.

pbatard, 2019-01-03:
> I can successfully UEFI boot using standard UEFI boot on USB (using fat32)

That doesn't mean anything. Maybe they screwed up FAT16/FAT12 support, maybe they screwed up partition support (can only boot from first partition, but not last), or fail to ignore capitalization, or it could very much be a misconfiguration issue on your side. Are you __absolutely__ certain you disabled Secure Boot on your side? For instance, if you create a FAT32 drive and copy the UEFI Shell (or any other x64 UEFI application for that matter) onto it as `\efi\boot\bootx64.efi`, can you actually run that?

brookspeppin, 2019-01-04:
Yes secure boot is off, you can see in my video. I did some more testing a found out the following:
1. Dell appears to have different implementations of NTFS boot support on various model and BIOS revs. I tried the same usb stick on a newer Dell E7490 and it booted UEFI:NTFS fine.
2. I tried your suggestion a made a FAT32 partition with the UEFI:NTFS files and tried it again on the other non-booting system (E7470) and it did successfully boot to UEFI:NTFS this time. I also made that FAT32 partition the "First" partition on the USB key rather than the 2nd (like you do it), but not sure if that really matters.

I think it would be awesome if you could update the tool to use FAT32 formatting for UEFI:NTFS to provide a wider range of support given the differences in hardware vendors' implementation. I get that might impact other boot support (linux, etc) so perhaps when an standard windows ISO is detected, it could do the fat32 partitioning and leave standard FAT for others. Thanks for all the help and response. Love your tool.

Also, I'm not sure if you are interested, but it isn't necessary to even use the unsigned UEFI:NTFS boot files when you have a Windows 10 Setup media. You can instead put the boot files from the media on that FAT32 partition (make it around 500mb or so) and then put the sources folder (as the install.wim is often larger than 4gb) on the nfts partition. This allows you to keep secure boot on and not have to worry about unsigned NTFS bootloader. I've got this working in a batch file and it works great. Every Dell system (as well as VMs) boot to it successfully with secure boot on. I've attached the batch file in .txt format. 
[usbcreate-v2.txt](https://github.com/pbatard/uefi-ntfs/files/2727247/usbcreate-v2.txt)

pbatard, 2019-01-04:
> Dell appears to have different implementations of NTFS boot support on various model and BIOS revs.

You do realize that if Dell implements NTFS boot then you __don't__ need UEFI:NTFS at all. UEFI:NTFS is used to compensate for UEFI firmwares that don't have a native NTFS driver. Obviously if the firmware has native NTFS support, then you will be able to boot the NTFS partition, and can select that one instead of UEFI:NTFS.

> made a FAT32 partition with the UEFI:NTFS files and tried it again on the other non-booting system (E7470) and it did successfully boot to UEFI:NTFS this time.

So this is a pure Dell firmware bug.

> I think it would be awesome if you could update the tool to use FAT32 formatting for UEFI:NTFS to provide a wider range of support given the differences in hardware vendors' implementation.

Can't use FAT32 for a small partition. That's why we use FAT12. But really, if the Dell firmware was implemented properly, it shouldn't matter if FAT12 or FAT32 are used, or if the FAT partition is the first or last. Also, please understand that the reason we use FAT12 in the first place and a partition that is last on the drive is because:
* On most Windows system (and especially Windows 7), you can __NOT__ mount 2 partitions at the same time from a removable drive. So we wouldn't be able to format one partition (NTFS one) and then the other (FAT one) without having to go through a cumbersome mount/unmout process
* Again, because you can't get 2 partitions mounted from removable drives on most Windows systems, we __must__ have the UEFI:NTFS partition at the end, so that when user plug/replug their drive, it's the first partition that they will see (NTFS one, with the ISO content) and not the UEFI:NTFS one, which would be useless to them.

So, to work around these limitation, we don't format the UEFI:NTFS partition at all, but instead, we simply write a pre-existing copy of this small partition, that is embedded in Rufus, at the end of the drive, which greatly simplify matters. And again, because that partition is so small (and of course we want it to be as small as possible), we can't use FAT32 for the file system but FAT12. But again, for a UEFI firmware that was designed by competent people, this should not matter, and I really have no interest adding workarounds because other people did a very poor job. So please get in touch with Dell, and tell them they need to fix their UEFI firmware.

> You can instead put the boot files from the media on that FAT32 partition (make it around 500mb or so) and then put the sources folder (as the install.wim is often larger than 4gb) on the nfts partition.

I am well aware of this. However, this is wasteful in space, and also, as I explained, this can't be used by Windows 7, Windows 8 and pre 1803 Windows 10 users, because, if you try to do that on removable media, you will only ever see the first 500MB partition when you plug the media, and be unable to access the rest of the drive.

Also, I'd like nothing more than submit UEFI:NTFS for Secure Boot signing (this is not a matter of money or unwillingness on my side), but the problem is that Microsoft are __abusing__ their power and preventing anything GPLv3 from being signed (for more on this, please see [this FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_do_I_need_to_disable_Secure_Boot_to_use_UEFINTFS)). So, I really don't see why I would have to completely change a solution that should work just fine with Secure Boot, just because the people in charge of Secure Boot signing feel entitled to say "Yeah, we don't __like__ the license you use, so we're not gonna let you help people that have Secure Boot enabled, even if your solution would work just fine for that situation too".

You also need to understand that, as I explain in the FAQ, __temporarily__ disabling Secure Boot is hardly the security liability that people seem to think it is (because surely, if it says "Secure" it shouldn't ever be disabled, right?).