pbatard/uefi-ntfs #20: Secure boot theoretically possible with custom keys

brozkeff, 2019-06-26:
I propose to modify the argument that Secure Boot cannot be used due to Microsoft lack of will to sign GPL3 code. While many (crappy) PCs do not have proper UEFI and have only the default (=Microsoft's) bunch of certified keys for Secure Boot, some better UEFIs (such as on my HP EliteDesk SFF PC) offer the user to actually push to the UEFI one's own signing keys, sign the bootloader etc with them, and have Secure Boot with ANY self-signed code. So while this option makes sense for only a very small minority of users, it is a legit option on the platforms that have a reasonable UEFI capable of replacing Microsoft's keys with alternative keys.

pbatard, 2019-06-27:
The issue here is that we (or at least the vast majority of people who use UEFI:NTFS) are attempting to boot Windows bootloaders, and it appears that Microsoft designed its bootloaders to also validate that they have been signed only with the Microsoft signature (which is very easy to validate during boot), and not a user-generated one.

See https://superuser.com/questions/1449698/secure-boot-custom-signed-bootloader for instance and especially this reply:
> You might be able to get a Linux OS to accept your own certificate but you will be unable to get Windows 10 to accept it. __There is no way to enable Secure Boot and boot to Windows 10 using your own certificate.__

Again, while I am well aware that, theoretically, one could add their own keys to the bootloaders and use _custom_ Secure Boot, in practice, this would be both very inconvenient for most users, who would have to go through what they see as a very obtuse and fairly technical process that they don't quite understand (hence fraught with user mistakes) and that wouldn't work when trying to install Windows anyway.

Ergo, I stand by my argument that _Secure Boot cannot be used due to Microsoft lack of will to sign GPL3 code._ because, unless you can __demonstrate__ otherwise and come up with a simple (and universal) process that even a non technical person can follow when they are trying to install Windows on a Secure Boot protected computer, what I say about Microsoft hampering the ability of people to use Secure Boot, even when most UEFI machines are supposed to make it __theoretically__ possible to do so, is the de-facto reality.

closed 2019-06-27 by pbatard

stdedos, 2019-09-01:
Hello there,

I'd like to point out that Microsoft is moving closer to Open Source (e.g. they recently published exFAT specs).

You might have more luck trying to tackle this

> As a developer, I'd like nothing better than be able to sign UEFI:NTFS for Secure Boot.
However, this is not possible because Microsoft have arbitrarily decided that they would not sign anything that is GPLv3 under the false pretence that it would force them to relinquish their private signing keys.

pbatard, 2019-09-01:
> e.g. they recently published exFAT specs

Yeah, about this: they did that... but only for the Linux kernel consumption.

Wanna add an exFAT driver to the EDK2 so that we can stop with the bullshit limitations UEFI firmwares have with FAT32? No dice.
Wanna add exFAT support to your Open Source app? No dice.

Once again, what we are seeing is that Microsoft are only "moving closer to Open Source" insofar as it directly benefits them.

If they were __actually__ moving closer to Open Source, they'd have made exFAT available for all Open Source applications. But if you read the terms & conditions under which they are making usage of their newly published specs available, you'll see that this is far from being the case...

Call me cynical, but I'll be convinced that Microsoft are actually "moving closer to Open Source" the day they drop point 4 from their [Secure Boot signing policy requirements](https://techcommunity.microsoft.com/t5/Windows-Hardware-Certification/Microsoft-UEFI-CA-Signing-policy-updates/ba-p/364828), because this __IS__ a blatant move against Open Source that, furthermore, can only be qualified as sheer abuse of power. Until that day people can try to point to Microsoft having become more Open Source friendly all they want, there'll still remain enough inconvenient facts to demonstrate that there remains a huge discrepancy between the image they are trying to present and what's actually happening at the ground level.

One also has to wonder, since NTFS has really become the de-facto file system to use for data exchange between heterogeneous systems where FAT won't do, why Microsoft are suddenly choosing to push for exFAT, instead of doing the one thing that would have been better suited to help Linux and other Open Source implementations, which is to make NTFS file system specs available for said implementations.

TL;DR: Even with their recent well publicized actions, colour me unconvinced that Microsoft are effectively "moving closer to Open Source".

stdedos, 2019-09-01:
> Call me cynical, [...]

No, please. That's the kind of answer I was expecting; I just didn't know "why" your answer was going to be this (i.e. what is the facade they are selling vs the reality).

I am not convinced with Microsoft's actions "for a very long time". However, it is hard to know the details of these issues, since I explicitly refuse even trying to do something productive on Windows (for exactly these reasons i.e. not hit stupid brick walls like that)

kttyb0f, 2020-12-05:
TIL the rufus guy is a crazy stallman-lite....

> Call me cynical, but I'll be convinced that Microsoft are actually "moving closer to Open Source" the day they drop point 4 from their Secure Boot signing policy requirements, because this IS a blatant move against Open Source that, furthermore, can only be qualified as sheer abuse of power.

have you actually read the gplv3? M$'s argument makes perfect sense. here's an excerpt from section 6

> “Installation Information” for a User Product means **any methods, procedures, authorization keys, or other information required to install and execute modified versions** of a covered work in that User Product from a modified version of its Corresponding Source. The information must suffice to ensure that the continued functioning of the modified object code is in no case prevented or interfered with solely because modification has been made.

so in some other schizophrenic rant against secure boot you posted "BUT THEY SIGN LINUX SHIMS!!!"

again, are you on your meds? are you doing okay? have you even read what license the SHIM is released under (spoilers: it's not GPLv3). guess why the SHIM exists in the first place? BECAUSE GRUB IS GPL. It had to get written to sidestep the secure boot signing requirement. so grub is not signed, now the shim is, the shim trusts a compiled-in certificate for any loaded binaries, problem solved. here's a link for you to actually read the shim, since you probably never have: https://github.com/rhboot/shim/

and there's nothing stopping you from taking a SHIM approach where you write a loader that validates the signatures of subsequent binaries against a hardcoded cert, getting that signed, and having the problem solved.

your M$ bashing is out of fashion and is just an excuse for laziness / incompetence. anyone can see that the GPL's requirement for disclosure of keys is retarded. cheers.

pbatard, 2020-12-06:
> have you actually read the gplv3?

I did.

> M$'s argument makes perfect sense.

Not even close. And I will be elaborating that at length below.

But I'm glad they managed to fool you, because it gives me a chance to explain explain how pernicious their "justification" that they are entitled to shut down the GPLv3 really is when trying to look at it logically...

> here's an excerpt from section 6
>
>> Installation Information” for a User Product means any methods, procedures, authorization keys, or other information required to install and execute modified versions of a covered work

And yet, without having to relinquish their keys, Microsoft does provide means for users to install and execute modified version of covered (i.e. GPLv3) work, since they have a procedure in place allegedly allowing anybody who wants to to sign their code for Secure Boot.

You'll notice that there's no mention of not having to pay or having to go through a potentially cumbersome procedure to get your GPLv3 code running, and of course, I don't think anybody (and by anybody, I really mean: the FSF) will argue (but feel free to do so if you want) that Microsoft should let GPLv3 code through that is deliberately designed to weaken the Secure Boot process. After all, Microsoft **already** have control of the signing keys, so if they believe that someone is going to attempt to use GPLv3 as a trojan horse to relinquish their keys or something else, they don't actually have to take pre-emptive measures, because, after all, they could just state, and this is the part that a lot of people, including yourself appear to have a lot of difficulty to grasp: _"We will place no condition on the license being used, but we do reserve the right to refuse service on a case by case basis"_.

There really was no reason to pre-emtively single out GPLv3 when they just had to add a line: _"Microsoft reserves the right to refuse to sign binaries for Secure Boot at their own discretion"_, which I'm pretty sure is already implied.

But instead, they floated the idea that GPLv3 would "magically" demand that Microsoft release their private signing keys, and that, the minute they sign a **SINGLE** GPLv3 binary for Secure Boot, there is no way, no appeal, no lengthy layers tractations that would save them from having to make their private Secure Boot signing key available to the world.

Ever seen a large corporation shake that much from trying to engage in a service where they could easily put legal terms of _"when using this service, our corporate rights trump your user rights"_? Yeah, me neither.

So, UNTIL PROVEN OTHERWISE, I can't see how one would buy the argument that has been put forward that the provision of a relatively open service for signing UEFI binaries for Secure Boot, which Microsoft does has in place, would run afoul of section 6 of the GPLv3. And that is not even taking into consideration the precedent of the SHIM in order to disprove the allegations that GPLv3 may force the relinquishing of private keys.

**Especially**, I have never heard a single FSF spokesperson even hint that _"Oh yeah, Microsoft are right to be cautious there, because, theoretically, this secction of tyhe GPLv3 could completely defeat Secure Boot, and also, there's certainly no way someone arguing in court that relinquishing the private keys or be forced to remove steps designed to keep a system secure, could ever win their argument that they should not be legally forced to take these steps, even if one could somehow interpret not-so-specific wording to imply as such"_.

As far as **I** am concerned (outside of the specific GPLv3 restrictions they added to their Secure Boot signing T&Cs), by providing an **open** service to sign binaries for Secure Boot, Microsoft does fit the requirements of providing "methods, procedures, authorization keys, or other information required to install (to) execute modified versions of a covered work". There's a procedure that anyone can follow to get UEFI bootloaders signed for Secure Boot, and get their modified versions of software work on a Secure Boot enabled machine, even without having to remove the Microsoft public key and installing their own (which, as long as no manufacturer of a computer prevents it -- and I can talk about the ARM Secure Boot fiasco/backlash if you want, which, I would assert, might have been the reason why Microsoft tried to weed out the GPLv3 for Secure Boot, as this is what they might have seen as a bigger challenge to their plan) I would also consider as more than good enough to satisfy the allegedly Secure Boot incompatible terms of GPLv3 section 6.

But again, I am not a layer, and unless we do see actual lawyers pronounce themselves on whether section 6 (or any other GPLv3 section for that matter) runs afoul of the current process that allows developers to sign their UEFI binaries for Secure Boot, I don't see how anyone can be 100% positive that it does, and that Microsoft are justified in their stance, which is my point.

Now, if you can point me to a quote from a legal person or the FSF, stating that, yes, as it stands, the Microsoft Secure Boot signing process is not enough to meet the requirements of the GPLv3, I will be more than happy to accept that Microsoft might be more atuned to interpreting legal licensing implications than I am, and that they had little choice but to place a wholesale ban on GPLv3 for Secure Boot... But I have now been waiting years for someone to do that, and so [... cut ...]
> anyone can see that the GPL's requirement for disclosure of keys is retarded. cheers.

Yet, the shim people at Red Hat don't seem to see it that way, since they happily sign GRUB UEFI bootloaders.

But it's nice to see how Microsoft managed to somehow convince people, like yourself, that the root of the problem has nothing to do with what, when looked at it closely and logically, is hard to construe as anything else but a purely arbitrary decision, and that "the GPLv3 terms are simply unworkable in terms of being compatible with modern security", which is great propaganda.