pbatard/uefi-ntfs #22: Booting 32bit iso's on 64bit CPU machines? "This system uses 64-bit x86 UEFI => searching for x64 EFI boot loader"

jddcef, 2020-11-27:
1) Write UEFI:NTFS to USB
2) Copy the contents of the 32-bit Windows 10 20H2 iso to the first partition.
3) Boot... 
but it fails

```
*** UEFI:NTFS (x64) **
[INFO] UEFI v2.50 (American Megatrends, Ox0005000C)
[INFO] American Megatrends Inc. X541NA.324 [INFO] ASUSTeK COMPUTER INC. X541NA
[INFO] Disconnecting potentially blocking drivers
(INFO) Searching for target partition on boot disk: (INFO PciRoot (0)/Pci(0x15,0x0)/Usb (0x1,0x0)
[INFO] Found NTFS target partition:
[INFO] PCiRoot (0)/Pci(0x15,Ox0)/Usb (0x1,0x0)/HD (1, MBR, OXC570CC50) [INFO] Checking if target partition needs the NTFS service
[INFO Starting NTFS partition service: CINFO1 Efifs NTFS driver v1.7 (GRUB 2.0)
[INFO) Opening target NTFS partition: INFO1 Volume label is 'ESD-ISO'
[INFO] This system uses 64-bit x86 UEFI => searching for x64 EFI boot loader
(FAIL] Could not locate 'efi\boot\bootx64.efi': (14) Not Found
Press any key to exit.
```
[image]

I was wondering, the "64-bit x86 UEFI" in the output, isn't that opposite?
and how do I get it to search for the 32bit efi file rather than the 64 bit efi file?

If I copy the 64bit iso contents, it boots fine on a X541N (which can't book from anything except this amazing booter).

pbatard, 2020-11-27:
I'm afraid you're trying to apply BIOS concepts to UEFI. It doesn't work that way.

Whereas BIOS (which is ALWAYS 32-bit) allows you to install either a 32-bit or 64-bit OS, UEFI only ever allows installing an OS for the same bitlength as the UEFI firmware.

This basically means that you either have a 32-bit or a 64-bit UEFI firmware to start with (this is decided by the manufacturer of your platform, and you can **not** change it), and therefore, if you install an OS in UEFI mode, you are constrained to only ever install the version of the OS that matches the bits of your UEFI firmware.

In your case, UEFI:NTFS reports that your platform has a 64-bit version of UEFI (and to answer your question, that's a 64-bit version of an x86 based UEFI, hence `64-bit x86 UEFI` as you could also have had a 64-bit version of an ARM based UEFI, in which case UEFI:NTFS would have displayed `64-bit ARM UEFI`. Oh and in case this is what you are asking about, x86 CPUs can be 32-bit or 64-bit (x86 does **NOT** equate 32-bit, despite what you may see elsewhere, and especially you should not use x86/x64 if you really want to mean x86 32-bit and x86 64-bit), so you **MUST** use a 64-bit version of an OS.

Especially, you will not be able to install Windows 32-bit in UEFI mode on that platform. And you certainly won't be able to run any 32-bit EFI binaries, because, as far as your UEFI firmware is concerned, it'f the same as if you were trying to run an ARM binary on an x86 system: it's just not compatible.

So, in summary, what you are trying to achieve will never work. if you have a 64-bit x86 UEFI firmware, you can't install 32-bit Windows 10 20H2 in UEFI mode. You have to install 64-bit Windows.

closed 2020-11-27 by pbatard

jddcef, 2020-11-27:
Thanks, I like your style. I learnt something here today. Thanks for taking the time to clarify.