pbatard/uefi-ntfs #23: How do I remove this?

liamfoneill, 2020-12-18:
I installed Windows with Rufus once which created the disk 'UEFI_NTFS' but have subsuently reinstalled Windows a number of times using the Windows Media Creation Tool (not Rufus). How do I safely remove this 512kb 'UEFI_NTFS'? 

[image]

closed 2020-12-18 by liamfoneill

pbatard, 2020-12-18:
The checklist is not optional. If you had followed it you would have found [this entry from the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Help_how_do_I_restore_my_USB_to_how_it_was_before_I_created_a_bootable_drive) that answers your question.

pbatard, 2020-12-18:
Ah, sorry, you logged this in uefi-ntfs where there is no checklist, rather than Rufus. But if you had an issue with what Rufus did, you should have created an issue under the Rufus project, not UEFI:NTFS, since it was Rufus that installed that partition.