pbatard/uefi-ntfs #26: Legacy boot seems to be non-functional after using this

RaXorX, 2021-07-09:
The USB boots fine in UEFI mode but legacy mode doesn't detect it as a bootable device. Is legacy booting not at all possible with this and is this purely intended to boot in UEFI mode only?

Masamune3210, 2021-07-09:
Considering the project is named uefi-ntfs, I would venture to say yeah, legacy boot is prob unsupported

pbatard, 2021-07-09:
> and is this purely intended to boot in UEFI mode only?

Yes.

Legacy does not need anything like this, because NTFS is supported by default from the MBR &rarr; VBR sequence of the Microsoft bootloaders.

Therefore, if you think you need a legacy version of this, I'm afraid that you are clearly misunderstanding whatever issue you think this would solve...

closed 2021-07-09 by pbatard

RaXorX, 2021-07-10:
@pbatard No, I wasn't saying I need a legacy version of this, what I was trying to say that legacy boot doesn't work if I make my USB UEFI:NTFS capable first. It boots fine in UEFI but not in legacy, not that I want a legacy version of this. I am not sure why it does so, pretty sure in the past it has worked fine for me but doesn't seem to be the case now.

pbatard, 2021-07-10:
> legacy boot doesn't work if I make my USB UEFI:NTFS capable first

Well, of course that won't work if you only apply UEFI:NTFS, because all you're doing when creating the UEFI:NTFS partition is taking care of UEFI boot. You are __not__ doing anything even remotely related to BIOS/Legacy boot if you are applying UEFI:NTFS. Again, BIOS boot and UEFI boot are completely orthogonal things, so you can't just solve one, and expect the other to magically work. You __must__ manually install something for each, if you want both UEFI and BIOS boot to work.

In this case, if you want BIOS boot to work alongside UEFI:NTFS, an in parallel to writing the UEFI:NTFS separate partition, you must:
1. Make sure your drive is partitioned as MBR, __not__ GPT (since GPT is incompatible with BIOS boot).
2. Install a Master Boot Record bootloader in the MBR (such as the Windows MBR bootloader) and make sure the volume you want to boot has a boot flag.
3. Install a Volume Boot Record bootloader in the NTFS partition (such as the Windows VBR bootloader, that looks for `bootmgr` and hands control over to it) or a secondary bootloader in the free space after the MBR (such as GRUB) or whatever your MBR bootloader code needs to complete the boot process.

In other words, if BIOS boot doesn't work, then you missed something from either 1, 2 or 3. But please understand that none of these have anything to do or are impacted by UEFI:NTFS. They're a completely isolated matter.

RaXorX, 2021-07-17:
Yes I realized that. My code of conduct was to use a aio-bootloader (just a grub based bootloader known as aioboot). Unfortunately I realized it was because it works without a boot flag and hence, when I used UEFI NTFS it marked the NTFS partition as bootable. Now, since the UEFI partition isn't boot marked, it still worked for UEFI. It would boot to the UEFI partition, hand it over to the ntfs selecting the boot files automatically and boot through it. 
Now when i booted in legacy, the problem occured because aioboot hates a partition marked as boot, seems the dev has put it that way for some reason. That's why it was not booting in legacy.
Removing the boot flag made it boot into legacy but then uefi stopped working. So it's not UEFI:NTFS's fault, I admit it there. I wonder if I could do something still though to make it boot into uefi if I removed the boot flag though.

Masamune3210, 2021-07-18:
Well the uefi itself doesnt care about the boot flag iirc, it just runs whatever efi file its told to load, as long as the uefi has the drivers for the filesystem the efi loader is on the rest is up to the loader itself

RaXorX, 2021-07-20:
@Masamune3210 That is weird then. Once i remove the boot flag it fails to boot for me in uefi. But then again I am using unconventional methods/bootloaders in the first place and not a direct bootable usb via rufus.