pbatard/uefi-ntfs #31: Makes the pendrive write-protected

AEJS879, 2022-03-06:


Masamune3210, 2022-03-06:
...sight. No it doesn't. It literally can't. If your drive switched to write protected it means that the drive has failed. It does not mean that Rufus caused it. Flash isn't immortal, it wears out. When it wears out to a point the controller can no longer access it correctly the controller automatically switches into write protect mode to avoid further damage or loss of data

AEJS879, 2022-03-06:
Rufus does make the partition on the pendrive write protected

Masamune3210, 2022-03-06:
No it doesn't. This is brought up so much it's answered in the FAQ

[image]

AEJS879, 2022-03-06:
but while writing on the pendrive rufus gives a write protection attribute to the partition 
[image]
see the attribute given to the partion 2

pbatard, 2022-03-06:
1. Please don't open duplicate issues with no details.
2. If you have an issue with Rufus, please report it in the [correct issue tracker](https://github.com/pbatard/rufus/issues) as you will find a check list that will help with the troubleshooting.
3. The 0x9000000000000000 attribute applies to the partition, not the drive, and it __NOT__ the source of your issue, whatever it is.
4. We still have no idea what your underlying problem is. If you are complaining about the fact that the UEFI:NTFS partition is write protected, this is intentional, because we don't want people to write there (since there isn't much space anyway). If you are complaining about the fact that the whole drive is write protected, then your latest post indicates that this is not the case, since Rufus is able to write to it. So, by all accounts, your report of _"Makes the drive write protected"_ doesn't seem correct.

I'm going to close this issue, as it's literally impossible to understand what you are reporting about, and you didn't open it in the proper repository anyway. The UEFI:NTFS project is not where you report issues about Rufus.

closed 2022-03-06 by pbatard