pbatard/uefi-ntfs #33: Cannot boot to syslinux.efi

pablocool, 2022-04-19:
I have created syslinux 64bit efi setup on fat32 partition and that boots fine. However I cannot make it work from NTFS partition using your uefi-ntfs on second fat32 partition. Uefi-ntfs just frozes on
[INFO] Launching 'efi\boot\bootx64.efi'...

Is it supposed to work with syslinux efi?

pbatard, 2022-04-19:
It's supposed to work with standard UEFI bootloaders.

Generally, when the last thing you see is `[INFO] Launching 'efi\boot\bootx64.efi'...`, the problem is with that bootloader and not UEFI:NTFS, since UEFI:NTFS has relinquished execution to that executable using the standard `LoadImage()`/`StartImage()` API calls.

My guess is that your Syslinux EFI bootloader has an issue, and you may have to add instrumentation to find precisely where that issue lies. For instance, it could very well be that this Syslinux bootloader is looking for its configuration file on a FAT partition, and is unable to proceed when it's located on an NTFS one.

I would also encourage you to find a machine that has native NTFS support in UEFI (more and more motherboards do) and see what happens when you boot Syslinux there. I'm pretty sure that you will see the same issue then, which will confirm that your problem has nothing to do with UEFI:NTFS.

pablocool, 2022-04-20:
Thank you for quick answer.
I agree that must be something wrong with Syslinux EFI itself. That is even more confusing, as Syslinux claims it has support for NTFS and Syslinux BIOS/MBR version boots from NTFS partition just fine.

Anyway I will try to find motherboard with native NTFS support and confirm that problem with Syslinux EFI has nothing to do with UEFI:NTFS.

pbatard, 2022-04-20:
Okay. I will close this issue for now then.

closed 2022-04-20 by pbatard