pbatard/uefi-ntfs #34: Booting Windows To Go from a usb with uefi-ntfs

jddcef, 2023-03-03:
Hi, I have a laptop (D330) that has a broken emmc drive on board, so only running an OS from the USB is an option. 
Ubuntu seems to boot fine, but WIndows To Go doesn't, so I want to use Uefi-Ntfs to boot it up instead as I find it works well on devices that just don't want to boot with any other method.

So, what would the steps be? 
Is it something like, write windows to go to another usb, and then copy the files to another uefi-ntfs formatted usb? But the Windows To Go usb seems to lack the boot/ folder etc

pbatard, 2023-03-03:
> So, what would the steps be?

1. Download Rufus from [here](https://rufus.ie)
2. Select a Windows ISO (or [download](https://github.com/pbatard/rufus/wiki/images/download.gif) it from within Rufus if needed)
3. Under _Boot selection_ select _Windows To Go_

Note that using UEFI:NTFS will **NOT** have any incidence as to whether _Windows To Go_ will work or not, as all it does is hand off to the Windows UEFI bootloader __which will be exactly the same__.

Therefore, unless your issue is with early boot and not being able to have your UEFI firmware executing the `/efi/boot/bootx64.efi` bootloader from the Windows partition, UEFI:NTFS will not help. And if that is indeed your issue, the method I described above, which will install UEFI:NTFS automatically if needed (but most likely won't, since it should create an ESP and install the Windows bootloaders there) should help.

closed 2023-03-03 by pbatard

jddcef, 2023-03-03:
I thought the same thing about the handoff, but once with a laptop, no other boot method worked on USB, it would load off the OS or the setup, but it would always freeze or just stuck. Until I booted the same image with UEFI:NTFS.  It was regarding windows.
I wrote about it here: https://superuser.com/a/1605528/35525

That's why when I have a case like this where I can't get into windows my instinct is to try UEFI:NTFS which worked on a machine which didn't want to boot any of many varieties of bootdisks (rufus, windows iso creator itself, refind, ventoy etc) Well it booted but just didn't get past the first blinking loading if I remember.