pbatard/uefi-ntfs #35: Consider documenting behavior on a local (SSD) disk

oliviercailloux, 2023-04-09:
I consider copying UTFI:NTFS to my fixed disk’s EFI system partition in order to be able to boot Windows from an NTFS partition on that same disk. Is this supported? All documentation I see only mention USB drives. Is there anything in the UTFI:NTFS software that makes it work only on removable drives or are fixed disks supported as well?

Relatedly, please document what happens when more than one NTFS partition exists on the disk (which holds true on my fixed disk but may also in principle occur on a removable drive, I suppose). Will it boot the first NTFS partition? Boot a random NTFS partition? Crash? Reformat everything and send nukes to random countries? I understand from the discussion in issue #12 that UTFI:NTFS (the version here) does not offer a configuration option to boot a specific partition, but this leaves the default behavior mysterious to me.

pbatard, 2023-04-09:
>  Is this supported?

No, as UEFI:NTFS is developed with the goal of being used with Rufus. But there's no reason why it shouldn't work, as, by design, it shouldn't matter if the disk is USB or not.

> Is there anything in the UTFI:NTFS software that makes it work only on removable drives or are fixed disks supported as well?

No. There's no filtering occuring. UEFI:NTFS will see the first NTFS partition from the disk where the UEFI:NTFS (FAT) partition and bootloader resides. So UEFI:NTFS doesn't care about fixed vs removable or USB vs HDD/SSD.

> Relatedly, please document what happens when more than one NTFS partition exists on the disk

Not going to. The code, which is fairly straightforward, is the only documentation you'll get as I don't have time to invest writing documentation for it. 

>  Will it boot the first NTFS partition?

UEFI:NTFS is designed for usage with Rufus and therefore for disks that contain only 2 partitions (the NTFS one and the UEFI:NTFS FAT one). What happens in case there are more than one NTFS partition is **undefined behaviour** and will remain that way, as I have not plan to alter the code to resolve what partition will be picked one way or the other (which really depends on the order in which the UEFI firmware decides to enumerate the partitions).

>  Boot a random NTFS partition? 

You should assume it will. Unless there's something in UEFI specs that guaranteed partition enumeration in the order they are listed in the GPT table (which may actually not be the order they occur on the disk).

> Crash?

No.

> Reformat everything and send nukes to random countries?

[Unless Matthew Broderick found a backdoor to that WOPR supercomputer you keep in your basement](https://en.wikipedia.org/wiki/WarGames), that shouldn't happen either...

> this leaves the default behavior mysterious to me.

This leaves the default behaviour mysterious to everyone, as this is not behaviour that the program is designed to solve. The program is meant to solve Rufus usage and Rufus usage only. If you want to use it in a different context, and since the code is Open Source, I can only advise you to fork your own version and make it work as you wish.

pbatard, 2023-04-12:
I'm going to consider that this question has been answered and close this issue.