pbatard/uefi-ntfs #41: Request to allow UEFI:NTFS to browse and chainload  /xxx/EFI/boot/bootx64.efi in case /EFI/boot/bootx64.efi not found in NTFS PTN1 of USB Drive.

devdevadev, 2023-12-31:
Hi Akeo...

Happy New Year....

First lots of thanks for providing UEFI:NTFS to us.  I use uefi-ntfs.img as partition image in Partition 3 of my USB Drive. and map UEFI_NTFS.imgPTNrep3 to PTN3 of my USB Drive to boot Windows Files in the root of NTFS Partition 1 of my USB Drive.

Most of the time we also keep our personal files in NTFS Partition 1 of our USB Drive. So it become a bit problamatic to keep Windows files in the root of USB Drive. along with other useful files.

It will be very useful if UEFI:NTFS allow use to browse and chainload /xxx/EFI/boot/bootx64.efi from NTFS Partition 1 of USB Drive in case /EFI/boot/bootx64.efi is not present in the NTFS Partition of USB Drive.

This way we can keep our Windows files within either a pre-specified FOLDER or any DESIRED folder in the root of USB Drive. This way we can also keep multiple Windows files in different FOLDERS in the root of NTFS Partition 1 of USB Drive.

/AIO/EFI/boot/bootx64.efi
OR
/Rufus/EFI/boot/bootx64.efi

You can allow UEFI:NTFS to look xxx (AIO or Rufus or or Akeo or ANY NAME you like) folder in the root of NTFS Partition 1 for chainloading .efi files to boot Windows.

It will be more useful if you can allow UEFI:NTFS to enumerate all folders in the root of NTFS Partition 1 that contains Windows files for chainloading /xxx/EFI/boot/bootx64.efi . This ways we can keep multiple Windows ISO files in dferent folders in the root of USB Drive. and this way UEFI:NTFS will work like a MULTIBOOT WINDOWS bootloader.  And this way UEFI:NTFS will allow us to boot either boot WIN10 files in WIN10 folder, and boot WIN11 files from WIN11 folder and so on.  and we don't have to format USB Drive again and again.

/WIN10/EFI/boot/bootx64.efi
/WIN11/EFI/boot/bootx64.efi
/WIN12/EFI/boot/bootx64.efi

and this way UEFI:NTFS allow us to browse and select any one .efi file to proceed for booting. and we can keep multiple Windows ISO in the same USB Drive. You can also provide this option to extract Windows ISO file in a specified folder in the NTFS Partition of the USB Drive using Rufus if you prefer.....

It will be very useful if UEFI:NTFS can provide this feature to browse and select desired bootx64.efi from NTFS Partition 1 of USB Drive.

Hope it will be possible to implement UEFI:NTFS in similiar way ??

Thanks you so much for creating UEFI:NTFS.....

Love you Akeo for the great development in USB Booting.....

Regards.....

pbatard, 2023-12-31:
1. UEFI:NTFS is not developed for your usage, but for the specific usage of booting a Windows installation media to install a single instance of Windows.
2. Because UEFI:NTFS needs to be signed, even if I agreed that the feature you propose might be useful to enough people (which I don't), I do not want to add any feature or code that is not strictly necessary because this can introduce vulnerabilities. That is why for instance, whereas I could try to request Microsoft to sign the R/W NTFS driver, the driver that I actually sign is read-only. This means that I __really__ don't want to add any extra code that is not needed for the prime usage of UEFI:NTFS.

As a result, I have to reject this feature request. It's way too specific for your specific usage and it could introduce vulnerabilities.

closed 2023-12-31 by pbatard

devdevadev, 2024-01-01:
Happly New Year....

Thanks for quick reply....and explaining the side effect of the requested feature.

Now I have to go with .cmd approach to selected to choose desired ISO out of multiple Windows ISOs in NTFS Partition 1 after booting from UEFI:NTFS from Partition 3 of USB Drive....

BTW, Thanks for the UEFI:NTFS development.......

Why you want to sign R/W NTFS Driver ? What exactly will be the use case of R/W NTFS Driver over read-only NTFS Driver ?

What will be extra benefit of R/W NTFS Driver in UEFI::NTFS / Rufus ?

Regards..