pbatard/uefi-ntfs #45: Win 10 usb not showing in uefi bootloader

2006Kartik, 2024-07-24:
I am using dell inspiron 5558 ... i3 4005u ...
And HDD of 512 gb with GPT windows 11 installed...
Using sandisk 32 gb...

Windows 10 latest downloaded on 24 july this year just today....

When i flashed the iso image of windows 10 in GPT format it is not showing in uefi bootloader ... I mean that created bootable usb is not showing in UEFI bootloader... so fix it 

I tried multiple versions and the old version 3 of rufus also ... 
I also tried to chose the fat32 option on rufus v3 but it throws an error of install.wim greater than 4 gb .... 

Please try to add any sollution for this ...

pbatard, 2024-07-24:
UEFI NTFS is fully UEFI compliant so either:

1. (Most likely) Your UEFI firmware boot options are not set to prioritize UEFI boot over CSM/Legacy boot, or there are other settings (such as USB not being enabled for boot) preventing the UEFI firmware from locating the UEFI bootloader or,
2. Your Dell UEFI firmware is not UEFI compliant, in which case you need to contact Dell.

At any rate, I am 100% confident that this is not a UEFI:NTFS issue. As a matter of fact, I am aware of some Dell platforms (sorry, can['t remember the models from the top of my head) were Dell screwed up their UEFI firmware and fixed this with a subsequent BIOS update.

pbatard, 2024-08-29:
Closing this issue as there is no further update.

closed 2024-08-29 by pbatard