pbatard/uefi-ntfs #46: Not booting the boot manager.

kristiandyan, 2024-07-24:
I've been stuck on this error for months. Any idea what could have happened? I removed all of the connection including ssd's rams and reset the bios. I need help 🥹
[image]

pbatard, 2024-07-24:
For a start, It'd help if you told the manufacturer, model of your PC as well as the BIOS version.

But really, once Windows bootmgr starts, the boot process is 100% in the hands of Microsoft, so my guess is an incompatibility between your specific hardware/BIOS and the Microsoft bootloader.

kristiandyan, 2024-07-28:
Thanks so much. I appreciate the time. I just installed windows using CMD with two USB 1 rufus and 1 ventoy which is really weird. Anyways, keep up the good work!!! 😊

pbatard, 2024-08-29:
Closing this issue, as there is no further update.

closed 2024-08-29 by pbatard